ivanalayan15/JuanFi #6: Insert Coin timer

Beta-Infinity, 2021-08-11:
If client forgot to click the the save and connect button or run out of time on the portal, after the dialog box closes the coins that are inserted added to the sales inventory but it wont add up time.

ivanalayan15, 2021-08-11:
please upload a video for this scenario

falafox2020, 2021-08-12:
Nodemcu from fresh reboot, the first client that insert coin but not press the save and connect button can't login due to no existing user in hotspot but in ide terminal the node run the script for creating a hotspot user.. also in admin panel, then said user will be added and the coins also..

But the second client do the same thing can login with no problem right after the countdown is reach..

Same as the 3rd, 4th

But when i tried to reboot again the node, same bug for the 1st client..

ivanalayan15, 2021-08-12:
ok thank you for the feedback, i will investigate this issue

ivanalayan15, 2021-08-14:
After investigation of this issue, seems its confusing to not do any autologin if user failed to click the save and connect, changes would be also do the autologin/connect of voucher when user failed to click the save and connect

closed 2021-08-14 by ivanalayan15