ivanalayan15/JuanFi #11: Request ako sir: Voucher Type

jamzkie, 2021-08-16:


ivanalayan15, 2021-08-16:
voucher type na po sya, at may extend time n din

jamzkie, 2021-08-16:
Paano po? Need paba NodemCU? Nabasa ko yung readme. Yung SCRIPT lang ba kailangan? No need na maagay sa scheduler?

jamzkie, 2021-08-16:
Extend time via coucher bayan? Or extend time via Coinslot... Via voucher sana. Thanks for the reply.

ivanalayan15, 2021-08-16:
ay d ko po kasi mavisualized yung sinasabi nyo n voucher type, you mean po hinde mguhuhlog sa coinslot ang customer? prng mgggenerate lng ng voucher doon sa nodemcu? parang taga control si nodemcu ng generation ng voucher instead sa direct mikrotik?

jamzkie, 2021-08-16:
What I mean sir is Voucher Build in MIKROTIK only. We can generate Vouchers in Mikrotik and Login it into portal just like mikhmon BUT with EXTEND time via VOUCHER. Pwede ba ganon sir?

ivanalayan15, 2021-08-16:
Medyo di ko pas gets sir, hahaha anyway what i can imagine is parang mgegenerate k ng voucher sa admin system? and eextend mo dn yung time sa admin system?

jamzkie, 2021-08-16:
Yes po sir. Yung tipong need lang MIKROTIK makakapag VOUCHER code kana, at may EXTEND Time kapa. Yung di sana kailangan ng NodemCU since start palang ng business

jamzkie, 2021-08-16:
Type voucher- VOUCHER1 may 2mins LOGIN
Type voucher- VOUCHER2 5mins EXTEND
result - VOUCHER2 7mins

jamzkie, 2021-08-16:
Pasensya po kung malito yung suggestion ko. Malibog kasi ako hehe. Salamat po sa time na basahin suggestion ko sir.

ivanalayan15, 2021-08-17:
parang d po applicable sa project n to yung voucher type n walang nodemcu since yung program po kasi inauupload natin sa nodemcu, like mikmon po yung mikmon nmn kasi is PC base so kailangan meron nghohost no program

jamzkie, 2021-08-17:
Ok po thank you. Keep up the nice project.

closed 2021-08-17 by jamzkie