ivanalayan15/JuanFi #19: Request: Telegram Sales Report

jamzkie, 2021-09-05:
Sana pwede mag add ng Telegram Sales Report sir. Salamat...

Ito po example credits kay sir Harold sa Arclem Community
[Mikrotik Telegram Sales Report arclem vers](https://drive.google.com/file/d/1P2DrN68DWX8KnDDztKxLEkYtDNtjFENd/view?usp=drivesdk)

charmedonesfjo, 2021-09-18:
Base on the link submitted... its more on Mikrotik Side SCRIPT sequences. You can try to experiment that scripts provided. the only tricky part is how and where to get a telegram API to embedded into mikrotik hotspot portal and the specific URL for fetch URL command.

ivanalayan15, 2021-09-24:
also we need to provide some extra data in our voucher column field like amount

charmedonesfjo, 2021-09-24:
try to look at this.... have you heard this dweet.io ? I run up with Jonathan Cutrer blogs about pushing mikrotik stats into dweet.io
out of curiousity i tried it ... in a way its good since it was FREE the only concern was the security. hope it was helpful
[Script_dweet.io_jcutter.md](https://github.com/ivanalayan15/JuanFi/files/7226823/Script_dweet.io_jcutter.md)

ivanalayan15, 2021-09-27:
Please make sure you change your login script, see the Readme file

[image]

ivanalayan15, 2021-09-27:
Login script:

### enable telegram notification, change from 0 to 1 if you want to enable telegram
:local enableTelegram 1;
###replace telegram token
:local telegramToken "2021159313:AAHEBoOLogYjLCpSwVeKPVmKKO4TIxa02vQ";
###replace telegram chat id / group id
:local chatId "136140248";
### enable Random MAC synchronizer
:local enableRandomMacSyncFix 0;
:local com [/ip hotspot user get [find name=$user] comment];
/ip hotspot user set comment="" $user;

:if ($com!="") do={
	:local sc [/sys scheduler find name=$user]; :if ($sc="") do={ :local a [/ip hotspot user get [find name=$user] limit-uptime]; :local validity [:pick $com 0 [:find $com ","]]; :local c ($validity); :local date [ /system clock get date]; /sys sch add name="$user" disable=no start-date=$date interval=$c on-event="/ip hotspot user remove [find name=$user]; /ip hotspot active remove [find user=$user]; /ip hotspot cookie remove [find user=$user]; /system sche remove [find name=$user]" policy=ftp,reboot,read,write,policy,test,password,sniff,sensitive,romon; :delay 2s; } else={ :local sint [/sys scheduler get $user interval]; :local validity [:pick $com  0 [:find $com ","]]; :if ( $validity!="" ) do={ /sys scheduler set $user interval ($sint+$validity); } };

	:local infoArray [:toarray [:pick $com ([:find $com ","]+1) [:len $com]]]
	
	:if ($enableTelegram=1) do={
		:local mac $"mac-address";
		:local totaltime [/ip hotspot user get [find name="$user"] limit-uptime];
		:local amt [:pick $infoArray 0];
		:local ext [:pick $infoArray 1];
		:local vendo [:pick $infoArray 2];
		/tool fetch url="https://api.telegram.org/bot$telegramToken/sendmessage?chat_id=$chatId&text=<<==========New Sales==========>> %0A Vendo: $vendo %0A Voucher: $user %0A IP: $address %0A MAC: $mac %0A Amount: $amt %0A Extended: $ext %0A Total Time: $totaltime %0A <<=============================>>" keep-result=no;
	}
};

:if ($enableRandomMacSyncFix=1) do={
	:local cmac $"mac-address"
	:foreach AU in=[/ip hotspot active find user="$username"] do={
	  :local amac [/ip hotspot active get $AU mac-address];
	  :if ($cmac!=$amac) do={  /ip hotspot active remove [/ip hotspot active find mac-address="$amac"]; }
	}
}

ivanalayan15, 2021-09-27:
[image]

closed 2021-09-27 by ivanalayan15