ivanalayan15/JuanFi #24: Juanfi Setup takes too long or never detected

smrtrnx, 2021-09-17:
Hi Ivan,

I am trying to use the .ino file other than the binary release files. I have noticed that when flashing .ino setup doesn't seem to show up in my wifi network. Yet but with binary it works juat fine.

Any advice?

ivanalayan15, 2021-09-17:
hello, did you upload the SIFSS image?

smrtrnx, 2021-09-17:
do you mean the data files **JuanFi-master**\**JuanFi-nodemcu**\**data** yes using sketch data upload

ivanalayan15, 2021-09-17:
yes, hmmm that is weird,is there any can you show me the screenshot for Serial Log

smrtrnx, 2021-09-17:
i'll try to provide one. anyway thank you for this wonderful opensource project of yours. keeps me busy on this pandemic :)

ivanalayan15, 2021-09-17:
thank you, this make me also always at home instead of going outside to hangout :-)

smrtrnx, 2021-09-17:
same here Ivan, cheers! lets get this project going

ivanalayan15, 2021-09-17:
@smrtrnx  i think i already know the issue, because the one in the .ino file is 45 seconds... and in the bin file still old which is 15 second only, it should show after 45 seconds

smrtrnx, 2021-09-17:
hmmm let me test that out changing it to 15000 on .ino

smrtrnx, 2021-09-17:
Fantastic! it works just fine right now. Thank you Ivan I've set it to 8 seconds :)

ivanalayan15, 2021-09-17:
nice hehe. i will close this ticket then

closed 2021-09-17 by ivanalayan15

thekailanihouse, 2022-02-20:
hi @ivanalayan15 , can i use this repo for WEMOS D1 R2?

what im done:

1. using all library what this repo needs
2. uploading /data/admin folder to data sketch

but im getting error like this
[image]