ivanalayan15/JuanFi #25: Assigning Static IP on esp8266

smrtrnx, 2021-09-17:
Hi maybe this helps lets try to assign a static ip address on the device. using the esp8266WiFi.h library. i think this stops mikrotik from assigning a dynamic ip address each time esp8266 disconnected from the server.

//Define Static IP Settings
IPAddress local_IP(192,168,5,253);
IPAddress gateway(192,168,5,1);
IPAddress subnet(255,255,255,0);
IPAddress primaryDNS(8,8,8,8);
IPAddress secondaryDNS(8,8,4,4);

ivanalayan15, 2021-09-17:
hello, currently the solution i have on this issue is not on the JuanFi setup its rather we do it on MT config, we can set a static ip for a specific mac address

smrtrnx, 2021-09-17:
yup, i've been testing the whole time using the same method but every time you unplugged  the vendo for some reason and the dhcp leases expires. it will get you back on the initial setup. on the other hand when it tries to connect it does not uses the assign static ip anymore but rather the dhcp serves lease issue another ip address on the device. :)

ivanalayan15, 2021-09-17:
really... that is really weird many actually using this kind of method already and its working fine, my vendo itself running 5 mos already using the same ip and its not really changing at all. may be your mac address is being change?

smrtrnx, 2021-09-17:
I am using laptop to for the initial setup. thus it does not generate randomized mac. have you tried unplugging your vendo like over night?

ivanalayan15, 2021-09-17:
yes, i tried and also days actually

smrtrnx, 2021-09-17:
I don't know what exactly happens here in my end tried both method the bin release file and the .ino. using bin file it works great the setup wifi ssid shows in an instant. but the .ino method does not.

I can only use the software for a period of time and I need to do the setup once and for all

smrtrnx, 2021-09-17:
Let me test this out if I won't be disconnected and do the initial setup since we've made changes in the WIFI_CONNECT_TIMEOUT. are you considering the suggestion of having the esp8266 a static ip? 

cheers

ivanalayan15, 2021-09-17:
im thing it actually, but it should be optional, this is usable also for LAN based actually so im considering it

smrtrnx, 2021-09-17:
Hmmm its been working lately but i get disconnected from the server, there must be something wrong with my router setup maybe. I don't have a clue right now. I can't connect to the router as of the moment

charmedonesfjo, 2021-09-18:
I been using the same method (DHCP then mark as STATIC at mikrotik side) of Ivan years ago without NO issue the ONLY different I've been notice is the wifi library... On Ivan's code HE uses <ESP8266WiFiMulti.h> while on mine is Wificlient.h.

charmedonesfjo, 2021-09-18:
[image]
Possible include an option for the users to choose a STATIC or DHCP

ivanalayan15, 2021-09-19:
@charmedonesfjo wow this is nice mock up, i will consider this for the enchacnement

bastewifi, 2021-09-20:
> [image]
> Possible include an option for the users to choose a STATIC or DHCP

nice mas oaky kung may ganito sana :)

smrtrnx, 2021-09-20:
Should we create a dashboard mock up for every possible functionalities to be added in the future this is a great idea. All necessary MT configuration should be done in single daahboard. What do you think?

ivanalayan15, 2021-09-21:
@smrtrnx much better to have mock up

btambalong, 2021-09-26:
> > [image]
> > Possible include an option for the users to choose a STATIC or DHCP
> 
> nice mas oaky kung may ganito sana :)

✓✓ nice..

ivanalayan15, 2021-09-30:
[image]

closed 2021-10-01 by ivanalayan15

btambalong, 2021-10-01:
bagong version po ba ito boss..

ivanalayan15, 2021-10-01:
@btambalong yes

btambalong, 2021-10-01:
thnx po boss..