ivanalayan15/JuanFi #28: Insert Coin button and Promo Rates not loading

smrtrnx, 2021-09-17:
Hello again Ivan,

on login.html the insert coin button doesn't seem to work while the promo rates button does not load its rates using Android Galaxy S9. Should this mobile compatibility issue?

cheers

ivanalayan15, 2021-09-17:
hi @smrtrnx , are you using the latest template, it should be solve by the latest template actually, even the captive portal is already solve the problem as long as you use the template in the master branch

ivanalayan15, 2021-09-17:
this is the exact fix actually on this issue

https://github.com/ivanalayan15/JuanFi/issues/22

smrtrnx, 2021-09-17:
Yes, I am actually using the master branch html templates. even tried the fixes tested on two mobile phones right now. don't have a luck.

ivanalayan15, 2021-09-17:
may i know if its a captive portal or a builtin browser?

smrtrnx, 2021-09-17:
Captive portal is the one that open automatically to the login page is that correct. Yes i tried that throws an insert coin error. On the other hand i tried chrome browser did the same thing

ivanalayan15, 2021-09-17:
yes that is the captive portal, i already tried it actually in very old phones that has very old browser and it works fine with it, i use many emulator actually to solve it

smrtrnx, 2021-09-17:
Was the html template on the master branch has its latest version of it right? I'll continue testing maybe i have mistakenly implemented it.

smrtrnx, 2021-09-17:
Here is what I found using the console on google chrome. 

192.168.xx.x/topUp:1 Failed to load resource: net::ERR_FAILED
192.168.xx.x/topUp:1 Failed to load resource: net::ERR_FAILED
﻿
the screenshot below points to core.js on line 148

[image]

[image]

​

smrtrnx, 2021-09-17:
Hi Ivan good day, I think i already know what's causing the insert coin not to work based on the error provided above. I just simply went to the router **firewall** > **filter rules** and disable the NodeMCUIP rule generated from the walled garden script. 

the filter rule mentioned above also blocked me from connecting to the router, this is in relation to my previous opened issue [https://github.com/ivanalayan15/JuanFi/issues/25](url) disabling the filter rule generated from the walled garden script will also fix this problem.

all good for now. insert coin is working very well.

ivanalayan15, 2021-09-18:
@smrtrnx thats goods to here, actually i notice that one actually before its weird right why the filter rule will work on some phones while in other is not it should have any difference with phone since its firewall level, anyway im closing this ticket great catch

closed 2021-09-18 by ivanalayan15