ivanalayan15/JuanFi #29: Issue: not mine(ctto)

jamzkie, 2021-09-18:
Napan sin ko sir sa JuanFi

1. Pag nagedit ka sa system config tas nagreboot. Ndi na xa connected manualy power off / onn

2.ung sa scheduler nia. Dindagdag ung current plus ung rate... Whic means ung 5php ko 2hrs pero validity nia 1day kung ada ako ng add nagdadag din ung validity nia sa scheduler.

3.ung s ap password di gumamana.. Pag nilalagyn ko ndi nkakak konek si node

Ps. For 24 hour test o o yan eto setu ko dun sa mga nag taning

Nilalagay ko nlng to sa github

smrtrnx, 2021-09-18:
1. Are you referring to the system config html file or the system config dashboard? 
- Normally the system will reboot after the configuration is saved.

2. Correct, that is how the system is designed. If you add 5 peso and you reached 50 your validity also increases. You can manually setup your preferred validity in minutes at the system config > rates dashboard. System dashboard will only appear once it is saved but will only appear if  it cant establish connection from your router.

3. Please check if you have entered the correct wifi ssid and password.

SemiTok2, 2021-09-18:
1. dashboard on nodemcu.It seem not realy reboted.LED indicator not working but idk if i use LED lights but im using RGB LED 3 wire Running and Blingking.
2. yes. . .may it should only add da time not the validity itself.if i ad time it will also add the validity.
3. yes its just 123456 and it wont connect and another if i hide SSID it wont also connect.

jamzkie, 2021-09-18:
> 1. dashboard on nodemcu.It seem not realy reboted.LED indicator not working but idk if i use LED lights but im using RGB LED 3 wire Running and Blingking.
> 2. yes. . .it should only add da time not the validity itself.if i ad time it will also add the validity.
> 3. yes its just 123456 and it wont connect and another if i hide SSID it wont also connect.

Are you the owner of this issue/post? I just stole it on fb and posted it in here.
https://www.facebook.com/groups/1172413279934139/permalink/1178379042670896/

ivanalayan15, 2021-09-20:
@jamzkie lan based ba ito?

ivanalayan15, 2021-09-28:
closing this issue

closed 2021-09-28 by ivanalayan15