ivanalayan15/JuanFi #30: Enhancement 💯 additional features

charmedonesfjo, 2021-09-18:
since the system doesn't use PROFILE account at the mikrotik side. Possible to ADD RX/TX (Que bandwidth management) at the RATES section AdminPORTAL SETUP. Sometimes there are vendo owner wants different SPEED Rates base on amount.

Also to Include ALL GPIO pin outs DROP DOWN LIST selection.

ivanalayan15, 2021-09-19:
hi @charmedonesfjo thank you for the suggestion,  i did not include some GPIO there because its reserve porpusely for LCD integration so thats why you can see its being hide. or are you referring to other GPIO pins like analog pins? cheers

ivanalayan15, 2021-09-24:
@charmedonesfjo do you have any script can be use for this

charmedonesfjo, 2021-09-24:
> 
> 
> @charmedonesfjo do you have any script can be use for this

@ivanalayan15 yes I have a script I used it from other systems. Downside... It will triggered only from the comment section of the particular code. The Mikrotik vendo system they put the amount value on the hotspot users comment section from their I can ruled out which profile that code going to use. At JuanFi case you could use the email field to put the amount of customers total coin purchased in this format e.g 10@juanfi.me (amount@anyemailformat.xxx) 
[Script_FIXED_JNHL_ProfileRATES.md](https://github.com/ivanalayan15/JuanFi/files/7225550/Script_FIXED_JNHL_ProfileRATES.md)

ivanalayan15, 2021-09-24:
@charmedonesfjo this is really nice, got it thanks for the idea.

charmedonesfjo, 2021-09-24:
> 
> 
> @charmedonesfjo this is really nice, got it thanks for the idea.

btw, when you execute the telnet command should ADD like this format-->  email=amount_variable@juanfi.me

ivanalayan15, 2021-09-24:
hi @charmedonesfjo what i see in the script actually is its matching a particular amount going to a particular RATE profile, can we just add a user profile option in our Rate config in our admin to simplify it.

for example:

100 / 7 days , profile = VIPRATE

where VIPRATE is user profile so that no more matching in mikrotik script and it will direct injection?

charmedonesfjo, 2021-09-24:
The script i posted still needs tobe modify
and if you will sticked to one profile setup at the nodemcu admin portal.

That will be the best approached at the admin portal rates another field for individual profiles base on rates.

Much easier if each profiles can be integrate/auto create upon saving will also write up at mikrotik device hotspot profile also. This is the sequence where other player can't be done.

ivanalayan15, 2021-09-25:
added new field for custom profile of specific rate:

[image]

Logic: which ever the highest amount in the rate that is applicable for the user inserted coin that is the one will be use as user profile.

closed 2021-09-27 by ivanalayan15