ivanalayan15/JuanFi #32: Incorrect time generated by the voucher

smrtrnx, 2021-09-19:
Hi Ivan, 

I had encountered another weird stuff. When inserted 1 or 5 or 10 peso coin the time displays is in random its not displaying the correct time anymore. I never change any settings when its working well during the first few days of testing. I also reflashed nodemcu but the problem still there.

I have checked all the wiring and everything is correct. I had a doubt about the coin acceptor. have you also encountered the same issue?

this issue started when nodemcu freezes during insert coin process.

have some thoughts about this?

cheers!

ivanalayan15, 2021-09-19:
hi @smrtrnx seems its cause by no common ground wiring between your nodemcu and coinslot? also please check if the coinslot setting is set to FAST and NO

cheers

smrtrnx, 2021-09-19:
Good day Ivan, i missed the common ground :) thanks

closed 2021-09-19 by smrtrnx