ivanalayan15/JuanFi #37: Random Mac Synchronization feature

ivanalayan15, 2021-09-20:
Random Mac Synchronization feature

needed

EdSacis123, 2021-09-21:
needed talaga random mac fix dahil sakit sa ulo sa mga naka multi ap

charmedonesfjo, 2021-09-23:
I think the script needs an additional validation or verification base on users device. That additional information may be pulled out from Mikrotik DHCP server. That the primary role of the dhcp server to gather some info base on what kind of device try to connect into like hostname/device OS/models are some basic field.

ivanalayan15, 2021-09-23:
@charmedonesfjo thanks for this its very useful

smrtrnx, 2021-09-24:
What about 
https://globalreachtech.com/products/hotspot-2-0-wi-fi/
Can this be integrated

ivanalayan15, 2021-09-24:
Workaround: https://github.com/ivanalayan15/JuanFi/wiki/Random-Mac-Synchronization

smrtrnx, 2021-09-24:
Hi Ivan,

Authenticated users may enjoy unlimited internet :) without voucher code leaving the uptime limit steady. Can you please verify if it also happens in your end?

More power to you Ivan. You have done a lot for this project. Good Job! I'll post whatever I discover that needs improvement. I am having fun with it. Thanks for this.

Have a wonderful weekend

charmedonesfjo, 2021-09-24:
> Hi Ivan,
>
> Authenticated users may enjoy unlimited internet :) without voucher code leaving the uptime limit steady. Can you please verify if it also happens in your end?
>
> More power to you Ivan. You [...]

this is very rare issue would like to know the version of your RouterOS 
1. specific user/voucher or ALL accounts?
2. have you tried to checked specific user/voucher account if the LIMIT TAB then look at LIMIT UPTIME field had a value ? 
3. NO Sudden Power interruption ?

smrtrnx, 2021-09-24:
Tested on two vouchers. Users are logged in and not shown on the ACTIVE hotspot users. on the login portal no registered voucher code are shown while they can still browse the internet. I've change nothing on my MT settings except for the script embedded.

User is already expired while the other is not.

No power interruption.

Its a rarely issue and i'll do another test if I still have the same issue.

thank you

closed 2021-09-27 by ivanalayan15

EdSacis123, 2021-09-27:
whats the final fix for random mac issues?

ivanalayan15, 2021-09-28:
@EdSacis123 MT script