ivanalayan15/JuanFi #47: SETUP

bastewifi, 2021-09-21:
hello sir ivan. pede po ba ung sa "lan based" setup, tulad din sa wireless ung may lalabas na juanfi setup para dun nalang mag ssetup?? 💯

ivanalayan15, 2021-09-22:
@bastewifi yes, will consider this enchancement

Pnote79, 2022-06-28:
Permission to ask
Sorry from Indonesia to change currency ..What to Edit?
In my country the coin is 500,1000
with a 500 voucher for 10 minutes and so on. Thank you for the answer

masterworkx, 2023-06-13:
> hello sir ivan. pede po ba ung sa "lan based" setup, tulad din sa wireless ung may lalabas na juanfi setup para dun nalang mag ssetup?? 💯

for what sir kaya nga po nag bibigay ng IP or nilalagyan ng IP si ESP32 and ESP8266 or ginagawang static kasi yun yung need mo i-access for example ip ni nodemcu mo is 10.0.0.254/admin yan yung need mo i access sa browser

masterworkx, 2023-06-13:
> Permission to ask Sorry from Indonesia to change currency ..What to Edit? In my country the coin is 500,1000 with a 500 voucher for 10 minutes and so on. Thank you for the answer

used currency converter