ivanalayan15/JuanFi #48: Request: Timestamp on Voucher generated voucher code

smrtrnx, 2021-09-22:
Hi Ivan,

it would be nice if we add some time stamp on the generated voucher on the mikrotik users list. so that it would be a lot easier for us to trace voucher codes in case clients is having trouble connecting.

cheers!

ivanalayan15, 2021-09-22:
@smrtrnx seems this is not possible since we dont have extra fields to save it

charmedonesfjo, 2021-09-24:
> 
> 
> @smrtrnx seems this is not possible since we dont have extra fields to save it

it would be possible. using radius type of authentication but it would be available only on a mid-range model of mikrotik it was not advisable low end.

smrtrnx, 2021-10-03:
@charmedonesfjo This is very useful telnet command but radius I guess is not enabled in haplite? The said feature is added in previous update by using the telegram. thanks for the input :)

charmedonesfjo, 2021-10-03:
> @charmedonesfjo This is very useful telnet command but radius I guess is not enabled in haplite? The said feature is added in previous update by using the telegram. thanks for the input :) Yet it's nice to have the timestamp since Ivan added a feature called voucher generation. :)

@smrtrnx YES, As I mentioned that you need a mid-range model for hosting a RADIUS enable system. technically haplite can be configured as radius client.

smrtrnx, 2021-10-04:
@charmedonesfjo may I ask what mikrotik router are you using? Is hex good enough. I am planning to have hex but i have no idea how it goes.. can you recommend please? 

Thanks buddy

charmedonesfjo, 2021-10-04:
> @charmedonesfjo may I ask what mikrotik router are you using? Is hex good enough. I am planning to have hex but i have no idea how it goes.. can you recommend please? 
> 
> Thanks buddy

@smrtrnx hEX Gr3 / hexS  preferred for theatre venue size.
hAPac2 and hAPac3 storage size issue if dont want to use external storage. affordable midrange models like rb4011/rb1100ah  new model rb5009ug

Right now im using RB3011  at home.