ivanalayan15/JuanFi #51: Enhancement Idea(s): 10 Minutes Uptime Limit Notification

smrtrnx, 2021-09-24:
Hello Ivan,

I was thinking about giving the user a notification, 10 minutes prior to their uptime limit. Please consider this enhancement :)

Cheers!

ivanalayan15, 2021-09-24:
@smrtrnx seems this is a limitation, ideally the user browsing in his own which means he is playing games or doing facebook, so the browser which is the portal is open is not active so we wont able to do notification on it.

charmedonesfjo, 2021-09-24:
> 
> 
> @smrtrnx seems this is a limitation, ideally the user browsing in his own which means he is playing games or doing facebook, so the browser which is the portal is open is not active so we wont able to do notification on it.

I totally agree.... once the users are validated its hard to triggered such routine / Why ? its already Mikrotik side not nodemcu or login.html file. unless you can create a scheduled routine + SCRIPT that will run every 60secs to get all expiring users code then notify them.

ivanalayan15, 2021-09-24:
closing this issue @smrtrnx

closed 2021-09-24 by ivanalayan15