ivanalayan15/JuanFi #56: Generate issue

bastewifi, 2021-09-27:
hello po sir. pansin ko lang po pag tapos mo mag hulog tapos hindi mo pinundot ung save/connect hindi sya mag aauto generate ng code..

may lalabas na code sa portal pera hindi sya nag ccreate ng code sa Mikrotik..

salamat po

ivanalayan15, 2021-09-27:
@bastewifi anung version ito

bastewifi, 2021-09-27:
2.1 po.

bastewifi, 2021-09-27:
wireless

ivanalayan15, 2021-09-27:
hmmm oks nmn s version ko

ivanalayan15, 2021-09-27:
@bastewifi sa extend time b sya or doon sa login n insert coin?

bastewifi, 2021-09-27:
hmm try nyo po mag insertcoin tapos hulog ka wag mo click ung connect wait mo po matapos ung countdown kung mag cconnect po sayo..

bastewifi, 2021-09-27:
> @bastewifi sa extend time b sya or doon sa login n insert coin?

sa mismong login po..

ivanalayan15, 2021-09-27:
@bastewifi yep triny ko sya ngayon oks nmn sya

bastewifi, 2021-09-27:
2.1 din po gamit nyo?? kasi nag test ako ulit ngaun same padin issue ee.

ivanalayan15, 2021-09-27:
2.2 gamit ko ngayon pero wala ako changes s logic n yun

ivanalayan15, 2021-09-27:
@bastewifi anyway baste i will release tonight the v2.2

bastewifi, 2021-09-27:
thanks sir

ivanalayan15, 2021-09-27:
this is not happening  on v2.2 closing the issue

closed 2021-09-27 by ivanalayan15