ivanalayan15/JuanFi #60: Enhancement... Hotspot Radius Authentication Capabilities

charmedonesfjo, 2021-09-28:
**For Centralized with multi RATES vendo. (Available only using HEXGr3 and Higher Models)** 

3 to 5 vendo (P2P or FOC connections)
Compound A. HEX Main 
Compound B. haplite (high rates) 
Compound C. haplite (rates same with B) 
Compound D. Coinslot Only (no mikrotik)
Compound E. haplite (low rates) --- > many competitor in the area

SCENARIO of using RADIUS enable system.
Users on A/B/C/D can crossover on each other using unified voucher code. They can used Wifi on other areas as long as it was connected on the same VENDO network provider/owner  except for USERS/Customer who transact on Compound E. ALL transactions made at Compound E can NOT use their voucher codes to Compound A/B/C/D 

haplite on Compound B and C may serve other purpose like VLANs / PPPoe etc. 

Similar approached to Globe GoWiFi when you avail Prepaid Data Promos bundled with Gowifi (available only mostly at affiliated Malls and Cities.)  

**_What do you think? send your feedback._**

ivanalayan15, 2021-09-29:
@charmedonesfjo hmmm this is more on enterprise setup of vendo, i have not tried yet this setup actually

charmedonesfjo, 2021-09-29:
that setup already UP and Running using voucher type via P2P network...

smrtrnx, 2021-09-29:
Thanks for the latest update Ivan. I have multiple AP setup. Surely authenticated users can crossover AP's as long as they have the same SSID authenticated users will be automatically connected. 

You may also want to make your AP ip address to be static as well as you declare it on the IP bindings.

But the idea of radius authentication is great as well and may cost us too much for the router?

Cheers

ivanalayan15, 2021-09-29:
@charmedonesfjo ok.... let me try this setup once i got the hex, so need more than 2 device on it?