ivanalayan15/JuanFi #66: JuanFI setup routine change

ivanalayan15, 2021-09-30:
JuanFI setup routine change

-when first time setup, no need to try to connect to SSID that has been setup, go to JuanFi Setup directly

ivanalayan15, 2021-09-30:
when setup is done, (once you save setting in system configuration) connection to AP will up to 180 sec until it will go to JuanFI setup again

smrtrnx, 2021-09-30:
Hi Ivan, 

Can we add a hidden ssid on esp where we can access admin panel anytime by typing the the static ip address assigned on esp? WiFiManager library may help. 

Cheers!

ivanalayan15, 2021-09-30:
@smrtrnx hi @smrtrnx  what do you mean hidden ssid?  because when you already connect to ssid you cannot host any AP anymore

charmedonesfjo, 2021-09-30:
> Hi Ivan,
> 
> Can we add a hidden ssid on esp where we can access admin panel anytime by typing the the static ip address assigned on esp? WiFiManager library may help.
> 
> Cheers!

I think i'ved already tried this. if there's no recent changes on the LIBRARY its not doable. 
Scenario : At first the SSID on AP/Router is on Broadcast Mode (able to scan the actual SSID name)
after I connect the NodeMCU module at AP in broadcast mode can connect to it....then I tried to hide the SSID from the AP then
reboot both device... after that connection failed. NodeMCU can't connect to AP with Hidden SSID.