ivanalayan15/JuanFi #69: wifi hide

bastewifi, 2021-09-30:
hi sir, pede bang maka konek ung node kahit naka hide ung WIFI name ng kinokonekan nya??

ivanalayan15, 2021-09-30:
@bastewifi hi, im not sure actually if its possible can you try it kung makakacoonect?

bastewifi, 2021-09-30:
hindi sya nakaka konek sir ee..meron kasi ung ibang system na ganitong setup lalu na sa wireless. kasi madalas na pag bitaw ng node.. kaya ginagawa ng iba ni hhide nalang ung SSId para solo lang ung node naka konek

charmedonesfjo, 2021-10-02:
> hindi sya nakaka konek sir ee..meron kasi ung ibang system na ganitong setup lalu na sa wireless. kasi madalas na pag bitaw ng node.. kaya ginagawa ng iba ni hhide nalang ung SSId para solo lang ung node naka konek

@bastewifi  can you specify the system in particular.... I've tried that approach you've mentioned several times using arduino sketch with ESP8266 but NO LUCK.  Try it your self with very basic sample sketch using wifi example libraries. let me know if its successful.

ivanalayan15, 2021-10-02:
@bastewifi may i know anung system to

ivanalayan15, 2021-10-12:
closing this issue since its really working even if ssid is hide

closed 2021-10-12 by ivanalayan15