ivanalayan15/JuanFi #77: Option for (Coin slot expired)

Jusher13, 2021-10-02:
Option for (Coin slot expired) 
1. kung mag o-auto login ba si vendo 
or 
2. ma sa-save lang yung credit nya

Meron po kasing pangyayari sir na yung mga coinslot ay namimili ng coin, kaya yung mga user ay nag papalit pa ng coins para panghulog at para hindi mag kulang ang ehohulog nilang coins. 
Guto po kasi ng iba na mag hulog ng 20 pesos, pero minsan pumapalya ang coinslot at na abotanan na sila ng count down, hindi na nila makukuha yung time ng rate20 kasi ng autologin na ang system.

Dito naman po sa v2.5-ota hindi nag sav-save ang credit sa portal once na naghulog ako at eca-cancel ko yung insertcoin countdown.
Nawawala naman yung modal ng insertcoin countdown pero hindi parin nahihinto ang countdown sa system.
Pag natapos na ang countdown sa system at pipindotin ko nanaman ang insert coin. Wala nang naka save na credit, back to zero po sya.

Pero nag gegenerate naman po sya ng code at time sa MT pag natapos na ang countdown ng system kahit hindi sya nag o-auto login dito.
Ito lang po sa tingin ko nakita kung problema sir ang hindi ma kuhakuha ni user ang 20 peso rate pag na abotan sya ng coutdown dahil nag o-auto login or generate ng code/addtime.

Maraming salamat po sir. More power :) .

charmedonesfjo, 2021-10-03:
> Option for (Coin slot expired)
>
> 1. kung mag o-auto login ba si vendo
>    or
> 2. ma sa-save lang yung credit nya
>
> Meron po kasing pangyayari sir na yung mga coinslot ay namimili ng coin, kaya yung mga [...]

@Jusher13 How about the EXTEND TIME routine ? still not sufficient solution for this scenario ?

smrtrnx, 2021-10-03:
You may want to re-calibrate your coinslot with a variety of coins.

closed 2022-09-04 by ivanalayan15