ivanalayan15/JuanFi #78: request: ESP32/Nodemcu Handle auto logout/login

RMel09, 2021-10-04:
Ganito po sa mga paid system once nghulog at bigla nalowbat or naclose ang browser automatic iforce login sya ni nodemcu/esp32, sa juanfi kc pag nakaclose ang login portal hnde nag auto login, kapag sa addtime hnde sya mag dadagdag unless ilogout at ilogin manually

smrtrnx, 2021-10-04:
@ivanalayan15 Hello Ivan,

How are you doing today?

This must be a client side issue, maybe we should implement such  thing called client side storage. kindly check this reference https://developer.mozilla.org/en-US/docs/Learn/JavaScript/Client-side_web_APIs/Client-side_storage

I hope this will help resolve the problem.

Have a wonderful day :)

smrtrnx, 2021-10-07:
my bad, localStorage is already implemented :( I apologize never noticed it on the corejs :)

closed 2021-10-16 by ivanalayan15