ivanalayan15/JuanFi #92: Script Modification Request

ElvinSDagohoy, 2021-10-15:
Thank you very much Master Ivan for the open source script if WiFi Vendo System. Moreover, may I suggest the following improvements;

1. If possible to add a script to a button assignment that will reset the monthly income according to voucher profile. Main purpose is to reset sales immediately during coins out 
2. Rates variable config if possible on Mikrotik as a script per Voucher Profile
3.  I have my script of multi vendo  sales report via email which I can share if needed.

Again thank you very much.

ElvinSDagohoy, 2021-10-15:
Another if possible is to set a variable schedule of Sleep and Wake up time of ESP8266 or ESP32 and LCD together sample every 2-3AM everyday

smrtrnx, 2021-10-15:
@ElvinSDagohoy I think you can achieve that in Mikrotik Scheduler.

Cheers!

closed 2021-11-08 by ivanalayan15