ivanalayan15/JuanFi #115: Buzzer Pin

cyberprime008, 2021-11-28:
pwede po bang my Buzzer pin

mrr3gx, 2022-01-21:
Yes ito sana ung irerequest ko. sana merong buzzer pin. like kung twing may maghuhulog magbabuzz.

ivanalayan15, 2022-01-21:
merong gnung coinslot n nabbili

iokdss, 2022-03-25:
Baka ang ibig nyang sabihin sir na buzzer hindi sa coinslot counter, Yung mag beep ang vendo sa countdown ng coinslot insert wait time countdown. Mag beep every seconds po. Pag press ng done mag generae ng code. Then mag start ulit yung countdown pero di na mag beep mag beep ulit pag 10 seconds nlng natira sa countdown mag beep sya ulit hanggang matapos ang 10 seconds to aware the client na dapat mag take na sya ng photo sa generated code sa LCD or to screenshot na code before mawala po. Parang countdown ng timebomb po. ;)

ivanalayan15, 2023-10-31:
Already implemented

closed 2023-10-31 by ivanalayan15