ivanalayan15/JuanFi #118: Soft WDT Reset

thekailanihouse, 2022-02-21:
Hi Mr. Ivan, I got issue with manual install from Juanfi-nodemcu.ino

1. I have already follow the instruction
2. after compile and upload code I got error with Soft WDT reset
Like this
`Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffbe0 end: 3fffffc0 offset: 01a0
3ffffd80:  00000000 00000001 3ffffe60 40212f94  
3ffffd90:  3fff25b4 3ffffde0 feefeffe 4021ad80  
3ffffda0:  00000009 00000100 3fff2270 3ffffe00  
3ffffdb0:  3162304d 3ffffde0 3ffffe60 40213070  
3ffffdc0:  3162304d 3ffe920c 3ffffe60 3ffffe00  
3ffffdd0:  3162304d 3ffffe60 000856a0 40201788  
3ffffde0:  00000000 3ffe8df2 01000000 40211c4c  
3ffffdf0:  31623034 3ffe8df2 3fff0984 40203084  
3ffffe00:  ffffffff 00000000 40217e40 3fff1e34  
3ffffe10:  00000000 00000100 feefeffe feefeffe  
3ffffe20:  feefeffe feefeffe feefeffe 001f8000  
3ffffe30:  3fff18e4 00000000 00000000 40218a20  
3ffffe40:  00000000 00000000 3ffffe56 3ffffe50  
3ffffe50:  0000001e 00fc0429 feefeffe feefeffe  
3ffffe60:  3fff25b4 9a3f9a3f 8a00ffff 00000100  
3ffffe70:  feefeffe feefeffe feefeffe 3fff2034  
3ffffe80:  00000020 000000fc 3fff18e4 402172c0  
3ffffe90:  3ffffed0 00000000 3fff1d34 000000f0  
3ffffea0:  feefeffe feefeffe 00000578 00000578  
3ffffeb0:  000000f0 3fff18e4 3fff18dc 40215d75  
3ffffec0:  00000578 4021e2fc 0000001c 40216832  
3ffffed0:  402153ec 402153c4 402153d8 001fa000  
3ffffee0:  00200000 00001000 00002000 00000100  
3ffffef0:  00000200 00000001 3fff0984 3fff0b50  
3fffff00:  4010047e 00000001 3fff1b2c 3fff0b50  
3fffff10:  3fffdad0 3ffffe00 3fff18dc 40215e0e  
3fffff20:  00002580 00000000 3fff0d84 3fff0b50  
3fffff30:  3fffdad0 00000000 3fff0984 40203b44  
3fffff40:  feefeffe feefeffe feefeffe feefeffe  
3fffff50:  feefeffe feefeffe feefeffe feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  feefeffe feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3fff0b50  
3fffffa0:  3fffdad0 00000000 3fff0b3c 40214094  
3fffffb0:  feefeffe feefeffe 3ffe85f8 40101315  
<<<stack<<<
`
this is decoder
`Decoding stack results
0x40212f94: String::concat(char const*, unsigned int) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\WString.cpp line 306
0x4021ad80: spiffs_phys_rd(spiffs*, u8_t, spiffs_file, u32_t, u32_t, u8_t*) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\spiffs\spiffs_cache.cpp line 167
0x40213070: String::concat(String const&) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/WString.h line 277
0x40201788: eeReadString(int, int) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/WString.h line 79
0x40211c4c: Print::println(int, int) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\Print.cpp line 238
0x40203084: populateSystemConfiguration() at D:\IoT\JuanFi-master\JuanFi-master\JuanFi-nodemcu/JuanFi-nodemcu.ino line 1349
0x40217e40: spiffs_obj_lu_scan_v(spiffs*, spiffs_obj_id, spiffs_block_ix, int, void const*, void*) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\spiffs\spiffs_nucleus.cpp line 332
0x40218a20: spiffs_obj_lu_scan(spiffs*) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\spiffs\spiffs_nucleus.cpp line 439
0x402172c0: SPIFFS_mount(spiffs*, spiffs_config*, u8_t*, u8_t*, u32_t, void*, u32_t, spiffs_check_callback) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\spiffs\spiffs_hydrogen.cpp line 138
0x40215d75: spiffs_impl::SPIFFSImpl::_tryMount() at c:\users\administrator\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\bits/unique_ptr.h line 665
0x4021e2fc: spiffs_impl::SPIFFSImpl::_check_cb(spiffs_check_type, spiffs_check_report, unsigned int, unsigned int) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/spiffs_api.h line 326
0x40216832: uart_init(int, int, int, int, int, size_t, bool) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\uart.cpp line 706
0x402153ec: spiffs_impl::SPIFFSImpl::spiffs_hal_read(unsigned int, unsigned int, unsigned char*) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/spiffs_api.h line 343
0x402153c4: spiffs_impl::SPIFFSImpl::spiffs_hal_write(unsigned int, unsigned int, unsigned char const*) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/spiffs_api.h line 337
0x402153d8: spiffs_impl::SPIFFSImpl::spiffs_hal_erase(unsigned int, unsigned int) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/spiffs_api.h line 340
0x4010047e: __wrap_spi_flash_read(uint32_t, uint32_t*, size_t) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_phy.cpp line 309
0x40215e0e: spiffs_impl::SPIFFSImpl::begin() at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/spiffs_api.h line 184
0x40203b44: setup() at D:\IoT\JuanFi-master\JuanFi-master\JuanFi-nodemcu/JuanFi-nodemcu.ino line 199
0x40214094: loop_wrapper() at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 198
`

what should i do?
this is what i've try

1. delete all value on system.data
2. delete file system.data
3. solve by commenting this line

`//detect if backup is exists
  int backupLength = eeGetInt(BACKUP_CONFIG_LENGTH_INDEX);
  if(backupLength > 0){
    Serial.print("Backup data found ");
    Serial.println(backupLength);
    String backupData = eeReadString(BACKUP_CONFIG_LENGTH_INDEX+5, backupLength);
    Serial.println(backupData);
    handleFileWrite("/admin/config/system.data", backupData);
    eeWriteInt(BACKUP_CONFIG_LENGTH_INDEX, 0);
    Serial.print("Backup data restored!, restarting....");
    ESP.restart();
    return;
  }`

thekailanihouse, 2022-02-23:
@ivanalayan15 please close this thread, my problem is solved bcos im not erase my flash bfore uploading script.. thank you

ivanalayan15, 2022-02-23:
ok got it closing the ticket....

closed 2022-02-23 by ivanalayan15