ivanalayan15/JuanFi #122: ESP32 Flashing using .ino

smrtrnx, 2022-03-19:
Hello Sir Ivan,

I've been trying to flash esp32 for a couple of days now but to no success, it's my first time using esp32 board. I have selected **esp32 dev module** from the IDE board manager.

when flashing I get lots of error codes. Can you please guide me the correct settings when flashing esp32. Please.. I looked back on the procedure but its only for the esp8266.

Best Regards

rexsllemel, 2022-04-23:
There is a binary file that is called JuanFi1.bin and JuanFi2.bin, flash then using pyflash not with arduino for less stress in flashing. You can then configure your esp32 later using JuanFi Setup wifi. You just have to download the latest released version. Unless you want to make your own program using those open source code.

smrtrnx, 2022-04-24:
@rexsllemel thank you for pointing this out. The binary should work. I never thought of it closing this thread now :)

closed 2022-04-24 by smrtrnx