ivanalayan15/JuanFi #123: Telegram just stopped working

smrtrnx, 2022-04-15:
Hi Sir Ivan,

Telegram just stopped working after using the latest version. Double checking my telegram credentials yet it was all correct. I also tried to revoke the API token still it doesn't work at all. I don't know how did this happen. I also tried rebooting the router after such changes on the hotspot user profile but to no avail.

Best Regards!

rexsllemel, 2022-04-23:
Every update there will be new portal along update. So after you'll done updating, you have to change your portal to latest version along with the downloaded latest file.

smrtrnx, 2022-04-24:
@rexsllemel I am using the version 3.3 template. It was working in a while after updating but later it stopped. I also tried creating new bots and group chat it does not work either.

smrtrnx, 2022-04-26:
Re-flashing it with version 3.3 and using the template version 3.3 does not resolve the issue. telegram still not working

smrtrnx, 2022-04-27:
Just an update ~ Telegram is working right now using the older version of the Hotspot Profile script. I don't have any idea why the latest script does not work with telegram in my end.

rexsllemel, 2022-05-19:
Hello, if you have version 4.0 you have to use version 4.0 template also. You may also have to configure Hotspot Users Profile.

smrtrnx, 2022-05-22:
Yes, I did that. I used version 3.3 software and with the 3.3 template version. I did configured the server profile as well pointing to **flash/hotspot** directory. Thank you :)

rexsllemel, 2022-05-23:
You may stay in 3.3 but, I was intrigued why it wasn't working on your end. If you want to update, you can reach help here :)

closed 2022-09-04 by ivanalayan15