ivanalayan15/JuanFi #127: Auto Eject coin if there is no request from system

SutrisnoIno, 2022-06-25:
Let me say thanks for your system that your made.

Now I use V3.3
But, there are something that I asking to you.
Can the system automatic eject coin if user not clicking "INSERT COIN" in web Portal Login page.

I was trying, user can insert coin whenever he's click or not "INSERT COIN" button on web portal login

Thanks,

smrtrnx, 2022-06-29:
Hi @SutrisnoIno that is not possible as of the moment, Juanfi system can only cancel INSERT COIN when the user accidentally pressed the insert coin option or does not wish to continue with the coin insertion.

The multi coin acceptor can always accept coins as programmed and will eject coins that is not programmed during setup. These are two separate systems(coin acceptor and Juanfi). Juanfi has no control over the multi coin acceptor not unless we can create or design a coin acceptor controlled by Juanfi system in the future.

Surely it will be so much fun if we can create one.

Regards,

SutrisnoIno, 2022-07-02:
Hi @smrtrnx,
Thanks for your answer.
But, can the esp8266 manage power to coin acceptor.
For example, if user press insert coin then esp8266 module turn on coin acceptor?

Thanks

smrtrnx, 2022-07-07:
Hello @SutrisnoIno 

coin acceptor runs on 12Volts meanwhile esp8266 is powered by 5Volts and can only distribute 3V through its designated pinouts.

Cheers!

glennford49, 2022-07-31:
I think this is posible using a transistor, the nodemcu pin will control the transistor switching with high/low, this transistor is connected to 12v source, coin acceptor and nodemcu

genmancoder, 2022-08-19:
you have to properly setup the set pin of the coin acceptor. It will automatically reject the coin unless the customer pressed the insert coin button in the portal.

jakecorn, 2022-08-26:
@genmancoder is right, you can enable/disable the coin acceptor using the SET pin. Just connect the SET pin (only the top pin or the red wire of the SET pin) to the nocemcu D0 or D5 pin then control the nodemcu pin either HIGH or LOW. D0 and D5 worked for me but you can also try the other pins.
```
int COINSLOT_PIN = D5;
pinMode(COINSLOT_PIN, OUTPUT);
if(coinSlotActive){
digitalWrite(COINSLOT_PIN, HIGH);       // accept coin
}else{
digitalWrite(COINSLOT_PIN, LOW);       // reject coin
}

```

closed 2022-09-04 by ivanalayan15