ivanalayan15/JuanFi #128: Issue Codes Missing

PisoWiFiTester, 2022-07-08:
If unused voucher code was entered in the insert coin portal. The code will automatically deleted resulting to invalid code or voucher not found. 
Ex. Si customer bumili ng voucher code ng hindi gumagamit ng vendo kumbaga nk print out n sa papel ung code. Eh first time ni customer gumamit. Ang ginawa nya pagkabigay sa kanya ng code imbes n i-type n nya directly sa portal un code ay ni press nya muna ung insert coin sa portal ska sya dun nag enter ng code. Ang resulta auto delete un code.  Or ganito po tlga nk design un portal sa Juanfi. Ty po.

JuanFi v4
Default Portal v4 rin po un ginamit.

ivanalayan15, 2022-07-08:
Thats how it was designed, Insert Coin means its a new transaction, Connect means use a voucher so you need to educate your user about it.

closed 2022-07-08 by ivanalayan15

PisoWiFiTester, 2022-07-08:
Thanks po Sir. for the confirmation po.