ivanalayan15/JuanFi #143: Step By Step Tutorial

masterworkx, 2023-06-13:
Good Day Boss Ivan wala naman po akong big issue sa JuanFi System siguro ang initial is setup or instruction nalang or pagkasunod sunod may iba kasi na kulang like sa Server Profile need pa kasi i Check yung HTTP CHAP, HTTP PAP, MAC COOKIES and COOKIES i think yan yung mga need nila kasi mostly jan sila nagkakaroon ng issue..

tulad nito gumawa ako ng initial procedure para maiwalan yung ibang mali
1. Flash NodeMCU
	- Flash bin1 file and Select "yes"
	- Flash bin2 file and select "no"
	- Get MAC Addres of ESP8266 (nodemcu)

2. Setup Mikrotik
	- Setup Interface
	- Setup Wireless (for hAPlite)
	- Setup Bridge HS
	- Add port on Bridge HS
		- ether2, ether3 and ether4
		- Add Wireless (for hAPlite)
	- Create IP Address for Bridge-HS
	- Create IP Address for LAN
	- Setup DHCP Server for Bridge-HS and LAN
	- Edit Pool for HS and LAN
	- Create DHCP Clients
	- Create ISP NAT
	- Create DNS and Check Allow Remote Reuqests
	- Setup Hotspot Server
		- Double click Hotspot Server on Addresses per MAC: change it to 1
		- Double click Server Profiles on Login Tab check this HTTP CHAP, HTTP PAP, MAC Cookies and Cookies
		- Edit User Profile and add the Scripts On Login and On Logout
			- This TOKEN: 6122145432:AAEpwahIVL19plH9TFgXqkVmGLi6tTV26vY - example
			- This ID: -800171458 - example
	- Create Mikrotik User
	
	
3. JuanFi Admin Setup
	- Connect to the WiFi of nodeMCU "JuanFi Setup"
	- Sytem Configuration Change all that you need
	- Connect nodeMCU on your Custom Board (CB) if you have
	
	FOR MIKROTIK
	- Check nodeMCU IP Address in IP>DHCP Server>Leases
	- Make Static the IP address of nodeMCU
	- IP>Hotspot Tab Double click the IP of nodemcu make binding then bypassed
	- Excute the Script (JuanFi Github)
		/ip hotspot walled-garden ip add action=accept disabled=no dst-address=10.0.0.254 <---nodeMCU IP Address
		/ip firewall filter add action=accept chain=input place-before=0 comment=NodeMCUIP src-address=10.0.0.254 <---nodeMCU IP Address
	- Go to hotspot folder Edit config.js located assets/js
		vendorIpAddress is your ESP8266/ESP32 (nodemcu) IP Address
		var vendorIpAddress = "10.0.0.254";
	- Upload Hotspot Templates
	- Copy the script for daily and monthly reset (JuanFi Github)
		/system scheduler add interval=1d name="Reset Daily Income" on-event="/system script set source=\"0\" todayincome " policy=ftp,reboot,read,write,policy,test,password,sniff,sensitive,romon start-date=Sep/28/2021 start-time=00:00:00;
		/system scheduler add interval=30d name="Reset Monthly Income" on-event="/system script set source=\"0\" monthlyincome " policy=ftp,reboot,read,write,policy,test,password,sniff,sensitive,romon start-date=Sep/28/2021 start-time=00:00:00;
		/system script add dont-require-permissions=no name=todayincome owner=admin policy=ftp,reboot,read,write,policy,test,password,sniff,sensitive,romon source="0";
		/system script add dont-require-permissions=no name= monthlyincome owner=admin policy=ftp,reboot,read,write,policy,test,password,sniff,sensitive,romon source="0"

	- Vendo Testing

ivanalayan15, 2023-06-15:
may be you can raise a PR for it?

masterworkx, 2023-06-26:
> may be you can raise a PR for it?

excuse Boss meaning of PR sorry for my noob question

EncEnl, 2024-02-06:
> @masterworkx PR means Pull Request para maisama yung tutorial mo sa mismong code repository

Google translate ftw. Remember that it's other then Philippinos using Juanfi 😂