ivanalayan15/JuanFi #145: OnLogin and OnLogout Script

smrtrnx, 2023-06-27:
I am testing out the updated  USER PROFILE onLogin and onLogout script it doesn't work in my end is there something that I am missing or it just not working at all.

every time that a new device login to hotspot, all active hotspot user will be disconnected. 

I am running version 4.3 on a 4.2 template version

@edmarlozada

edmarlozada, 2023-06-28:
> I am testing out the updated USER PROFILE onLogin and onLogout script it doesn't work in my end is there something that I am missing or it just not working at all.
>
> every time that a new device [...]

can you message us on facebook https://www.facebook.com/chloe.renae.9
we will get to the bottom of this issue. help us and we will help you too.

smrtrnx, 2023-06-29:
Hi, i don't think that the On Logout script is the same as before, or was it the revised script as the previous version? 

Can't find the 3.3 version of the logout script in here anymore.

Thanks for your response. More power

edmarlozada, 2023-06-29:
> Hi, i don't think that the On Logout script is the same as before, or was it the revised script as the previous version?
> 
> Can't find the 3.3 version of the logout script in here anymore.
> 
> Thanks for your response. More power

we found the bug that caused active user to logout! and your are right! we are sorry for putting doubts on you.
1. it was caused by the RandomMacSyncFix! and for now, we dont know why yet. we are still working on it.
2. if you are using the stable version of the mikrotik, you will be affected by this bug. but if you use the longterm, it will work normally.
we are working for the fix on that for now. thank you for your feedback. we will post the fix soon.
@smrtrnx

smrtrnx, 2023-06-29:
No worries buddy @edmarlozada, 

I forgot to mention in my previous post that I am running the latest software version of Mikrotik Hex.

Thank you for putting up some time. I appreciate your time and effort.

Cheer up :)

edmarlozada, 2023-06-30:
@smrtrnx @ivanalayan15 
here's the simple fix for the issue of All Active users being logout when a user login and when RandomMacSyncFix is set to 1 (enabled) and using the stable version of mikrotik.
- onLogin Script, on RandomMacSyncFix module. change the $user to $username! that's it!
  fr => :foreach AU in=[/ip hotspot active find user="$user"] do={
  to => :foreach AU in=[/ip hotspot active find user="$username"] do={

closed 2023-08-13 by smrtrnx