ivanalayan15/JuanFi #146: Version 4 missing

jariesdev, 2023-06-28:
Is version 4 already a close source or opensource? Can't see any version 4 related branch. Thank you

edmarlozada, 2023-06-30:
> Is version 4 already a close source or opensource? Can't see any version 4 related branch. Thank you
@jariesdev 
version 4 is here
https://juanfi.juansystems.com/

jariesdev, 2023-06-30:
@edmarlozada 
Sorry, I mean the source codes. I'm using Arduino to flash the device. The site provided compiled code already.

rexsllemel, 2023-08-24:
> @edmarlozada What I mean the source code for v4 since its claimed to be an opensource is not available on the repository. I'm using Arduino to flash the device. The site provided compiled code already so code cant be check.

Hi, source code for the latest version is not available. The developer of this project announced that the source code of the latest version will not be available because some developers copy his project and sell the program  which is against in the opensource community. I hope you'll understand. 
[image]

closed 2023-10-31 by ivanalayan15