ivanalayan15/JuanFi #150: Coin Acceptor

MeeeMaaaaw, 2023-08-13:
Hi, napansin ko lang, kapag nag-iinsert coin yung customer ko at na disconnect yung phone nya sa wi-fi hindi na naccredit yung ibang coins. Example Mag huhulog siya ng 5 pesos na tig pipiso then biglang nadiskonek yung cellphone nya sa Wi-Fi pero nahulog na niya yung pang Lima na coin, apat lang maccredit sa kanya. May fix ba tayo dito? kasi sa ibang PisoWiFi software kapag naghuhulog ang customer , at kahit pa madiskonek sila sa wifi na ccredit yung buong coins. Ang fix ko sa ganito ngayon is yung buy voucher nalang muna na gumagamit ng button at LCD. 

Madalas tong issue sa mga medyo lumang phone at sa mga mga iPad na lilipat na ssid dahil walang pang internet kapag nakakonek sa PisoWifi

smrtrnx, 2023-08-15:
Hello @MeeeMaaaaw,

Somehow I have encountered the same issue as you do. it was an android phone but not sure it was an older model of an android phone. keeps switching network / access point. 

For now i do not have a clue about the issue, since there is only one customer who is experiencing this kind of problem.

@ivanalayan15

rexsllemel, 2023-08-24:
> Hi, napansin ko lang, kapag nag-iinsert coin yung customer ko at na disconnect yung phone nya sa wi-fi hindi na naccredit yung ibang coins. Example Mag huhulog siya ng 5 pesos na tig pipiso then [...]

Hi, did you use the latest update? the latest version at the time of this writing is version 4.3. As far as I know, this issue has been fix on version 4.2 I believe. You can find the latest version [here](https://juanfi.juansystems.com/)