ivanalayan15/JuanFi #155: Issue on System Clock after upgrade to v4.3

mugnet, 2023-10-29:
Hi sir thank you sa opensource platform, issue ko lang po lagi kapag nag-upgrade ng 4.2 to 4.3 nagiging Jan 1, 1970 yung system clock. Naupdate ko na lahat sa mikrotik even NTP client and clock settings related to it.
Thank you.

MeeeMaaaaw, 2023-10-30:
Kindly disregard .. not a big deal issue. I think sa ROS na yata ang issue kasi before kahit 4.3 ang gamit ko tama naman ang time pero nung nagupdate yung ROS naeencounter ko na tong issue.

MeeeMaaaaw, 2023-10-30:
BTW reflashing will fix the issue but mga ilang months lang babalik lang yan ulit sa ganyan kaya i-disregard mo nalang yang issue sa time.

mugnet, 2023-10-31:
sa ROS nga boss, triny ko sa hexs v7 nagwork naman. Salamat sa sagot.

closed 2023-10-31 by mugnet