ivanalayan15/JuanFi #156: Juanfi manager voucher

Jmhwifi, 2023-11-21:
Hi sir, tungkol po sana sa juanfi manager kapag po nag generate ng voucher using juanfi manager  bakit po ganun hindi po ma maximize ung isang buong bond paper konti lang po ang kasya sa isang bond paper kapag nag print po ako sayang po kasi ung space ng paper na bakante sana po sir fix po salamat

ivanalayan15, 2023-11-22:
pde nyo po ireplace yung template sa settings yun nga lng need nyo edit ung html template para mafix yung issue sa mobile

Jmhwifi, 2023-11-22:
Ang hirap sir mag edit lalo na kapag walang alam sa html 😁pero try ko po hanapan ng magaling sa hmtl editing salamat sir

arisc007, 2024-02-28:
Hi po, message nyo po aq s edd.cordova13@gmail.com, send q po s email ninyo ang template n ginawa q. s 1 bond paper ay kasya upto 64-pc. voucher codes, n pwede ninyong i cut at ibenta s customers. pero codes lng po un at di po included ang bar codes.