ivanalayan15/JuanFi #159: Admin panel

ChassterTP, 2023-11-30:
No device ip address on admin panel dashboard

TataUdal, 2024-01-14:
How to pix it?
[image]

MeeeMaaaaw, 2024-01-30:
> No device ip address on admin panel dashboard

Please Provide Screenshot

> How to pix it? ![Screenshot_2024-01-14-09-28-44-86_cbe8c7c25be320d0c6f8847e2dea9fd0](https://private-user-images.githubusercontent.com/156556383/296525148-0ae80c60-ea6b-4147-b074-8067cd5b3114.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY1ODcwMjEsIm5iZiI6MTcwNjU4NjcyMSwicGF0aCI6Ii8xNTY1NTYzODMvMjk2NTI1MTQ4LTBhZTgwYzYwLWVhNmItNDE0Ny1iMDc0LTgwNjdjZDViMzExNC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEzMFQwMzUyMDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03Yjg5ZWRkMjI0YzA0MTExMDYzZTVhNGY5ZDkzNjkxNjlhMWU1ZmU4NjU3YTE3Y2JiMzU5NWJkODNlYzBjNWY5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.fkV9-t87hZNOEE0wd99BEMph8wq04nVjK_gxUyAVpBE)

Clear cache / data. Malamang nag update ka ng portal files nyan