ivanalayan15/JuanFi #163: Enable Lock to MacAddress Voucher

MeeeMaaaaw, 2024-02-12:
Sa mga naghahanap ng script para ma-lock sa MacAddress ng phone ang voucher, pwede niyo to i-add sa login Script

```
### enable Lock MAC - use 1 to enable
:local LockMac 0;

# LockMacAddress
:if ($LockMac=1) do={
:local mac $"mac-address";
/ip hotspot user set mac-address=$mac [find where name=$user];
}
```

arisc007, 2024-02-28:
Hi po, ask q lng po kung bkit need n i lock s mac address ng phone ang voucher? kung nk lock at nag change s random mac s paglipat s ibang AP, eh di, disconnected ang voucher?

MeeeMaaaaw, 2024-02-28:
Optional lang yan. May ibang owner kasi na need yan para di manakaw yung voucher ni client. Kapag nakaenable ang ranDomMacFix , pwede iLogin ng ibang user yung VoucherCode na gamit ng ibang client kung alam nila ang code

arisc007, 2024-03-03:
So, un pl ang reason, idol? Thanx for info.