ivanalayan15/JuanFi #167: Vendo machine

Aljebar, 2024-05-19:
Hello po.. ask lang po bakit po wlang interner ang vendo sa modem wifi po Meron.. ok naman po ang wire functional lahat

smrtrnx, 2024-05-19:
Hello,

1. Naka mikrotik router ba yung vendo?

2. If yes, check mo po yung dhcp server at hanapin mo po yung IP Address.

Gawin mo po static.

A.Select mo yung IP Address ng nodemcu,

B. Right click mouse at lalabas yung static ip address option

3. Pumunta ka sa IP>Hotspot>Walled Garden

A. At i add mo yung IP address ni nodemcu.

WALLED GARDEN ENTRIES:

A.1. SERVER: Hotspot
A.2: ACTION: Allow
A.3: DST Address: nodemcu IP address (ei. 10.0.0.2)

Aljebar, 2024-05-19:
Saan po check ko ang dhcp server ty. Po