ivanalayan15/JuanFi #173: Coin Slot pulse Hacked

jariesdev, 2024-07-30:
Hi, there. If I'm not mistaken, 1 piso = 1 pulse, 5 piso = 5 pulse etc. My issue encountered is when some static electricity detected like some device plugged to same outlet it detect 1 pulse. similar to just simply touching the ground of the device. Is someone encountered it, I know this is not software related but we can (i think) change the pulse setting to like 1 piso = 2pulse instead). Thanks!

raldincasidar-studio, 2024-08-21:
dapat may common grounds yung board mo at yung coinslot.

I mag connect ka din ng wire from coinslot ground papunta sa ground ng board

jariesdev, 2024-09-14:
Sinubukan ko ito sa bare setup. ESP32 at coinslot lang common ground na din ground sa ESP=Coinslot=power supplier pero kapag tap ko lang daliri ko sa ground nag register as +1 coin every time